RaccoonlabDev/mini_v2_node #50: May I ask if I can use your examples to develop a module that controls my CANopen servo motors using the Cyphal/DroneCAN interface?

zouzhe1, 2024-08-06:
May I ask if I can use your examples to develop a module that controls my CANopen servo motors using the Cyphal/DroneCAN interface?

Because I want to use ArduPilot or Pixhawk's rover to control my ground vehicle for waypoint navigation, and I am using DC permanent magnet synchronous motors, 48V, with CANopen protocol.

zouzhe1, 2024-08-07:
> Sure. It is possible with a mini v3 node (stm32g0) that has 2 CAN busses. I can provide an example by the end of this week.

@PonomarevDA     Thank you very much. Looking forward to your example

PonomarevDA, 2024-08-10:
Done. I've added an example with DroneCAN RawCommand on CAN1 and a custom CAN protocol on CAN2. More details you can be found in the PR: https://github.com/RaccoonlabDev/mini_v2_node/pull/54